nicoa47/LDEEG #50: first review round preprocessing

nicoa47, 2022-08-11:
Preprocessing for high density

1. notch filter (read first: https://doi.org/10.1016/j.neuron.2019.02.039 )
2. HP filter (cognitive analyses - use 0.5 Hz as in actionlab)
3. downsample to 100 Hz (common)
4. rASR
5. pyPREP
6. robust z scoring

nicoa47, 2022-08-12:
open question: which calibration data to use?

caghangir-1, 2022-08-14:
**Robust z-score normalization algorithm:**

 def robustZScore(self, raw):
        
        channel_amount = len(raw._data)
        raw_robustzscore = raw.copy()
        
        for i in range(channel_amount):
            MAD = np.median(np.abs(raw._data[i] - np.median(raw._data[i])))
            raw_robustzscore._data[i] = 0.6745 * (raw._data[i] - np.median(raw._data[i])) / MAD
            
        return raw_robustzscore

nicoa47, 2022-09-05:
since this is a general overview rather than a task, [moved to wiki](https://github.com/nicoa47/LDEEG/wiki/Final-decisions-about-preprocessing)

caghangir-1, 2022-09-22:
New smart tricks:
1) For high-density EEG, instead of Riemann ASR Euclid ASR will be used due to 144 times faster in computation and according to saccadic artifact reduction assessment, Euclid ASR with 'scf' estimator found to be highly effective to remove artefacts. The reason is that unlike rASR, Euclidian ASR could pass from the requirement of the regularization (You are obliged to use 'lwf' in rASR which is downside but still doing pretty amazing job).
2) Single file running for script with flag input addition. This hack RAM allocation problem since every single run only one file is preprocessed and kernel dies. All the scripts are run sequentially with fresh RAM allocation.

# ===== BCI ======
python very_new_preprocessing_applywholedata_v1.2_HPC 0; python very_new_preprocessing_applywholedata_v1.2_HPC 0;1;python very_new_preprocessing_applywholedata_v1.2_HPC 2;python very_new_preprocessing_applywholedata_v1.2_HPC 3;python very_new_preprocessing_applywholedata_v1.2_HPC 4;python very_new_preprocessing_applywholedata_v1.2_HPC 5
# ===== BCI ======

# ===== LD Cueing======
python very_new_preprocessing_applywholedata_v1.2_HPC 0; python very_new_preprocessing_applywholedata_v1.2_HPC 0;1
# ===== LD Cueing ======

# ===== Lucireta ======
python very_new_preprocessing_applywholedata_v1.2_HPC 0; python very_new_preprocessing_applywholedata_v1.2_HPC 0;1;python very_new_preprocessing_applywholedata_v1.2_HPC 2;python very_new_preprocessing_applywholedata_v1.2_HPC 3;python very_new_preprocessing_applywholedata_v1.2_HPC 4;python very_new_preprocessing_applywholedata_v1.2_HPC 5;python very_new_preprocessing_applywholedata_v1.2_HPC 6;python very_new_preprocessing_applywholedata_v1.2_HPC 7
# ===== Lucireta  ======

nicoa47, 2022-09-23:
Please move this [to wiki](https://github.com/nicoa47/LDEEG/wiki/Final-decisions-about-preprocessing) as explained.

closed 2022-09-23 by nicoa47