nicoa47/LDEEG #57: Task separations

caghangir-1, 2022-08-17:
**Cagatay:**
* Handling the preprocessing.
* Apply all the analyses except Beamforming.
* Write the method section, and formulas, create figures and write the analysis & results section with tables.
* Co-apply statistical assessments with Nico.
* Fine-polish conclusion

**Nico:**
* Write the introduction
* Write the discussion & limitations
* Find the type of analysis with the code in Python or at least make it clear for Cagatay to code it.
* Find statistical measurements of different types of analyses.
* Co-apply statistical assessments with Cagatay.
* Write the abstract
* Write the conclusion

caghangir-1, 2022-08-17:
**Discussion & limitations:**
* **Dilemma between sample size and EEG density level** (It influences the trust of the statistical results with a low sample size but also increases the trust using higher densities. This is called a statistical paradox. It could be a nice pump-up way to raise this concern in whole sleep science in the discussion).
* **Non-generalizability of the preprocessing and impossibilities regarding detection of overcleaning** (What do we know about the information processing of lucid dreaming? How do we sure about what are the evoked or event-related potentials behind the LD dream? It could be related to a subject's sub-conscious, current mindset, the dream content, and active\passive dream movements which are motor-cortical biases over more subtle information processing).
* **Different LD actions cancelling out the neural markers**
* **Different EEG devices with different hardware setups could generate variations among signals no matter how much we try to equalize them via applying robust z-score normalization.**
*

caghangir-1, 2022-08-28:
Introduction new addition:

* Micro-saccade literature reviewing and importance in REM sleep, how it influences creation artefacts, why they are essential to remove, and the limitations of removing them (related the only but the best journal review 4 different algorithms to detect micro-saccades, and point out the neural correlation of those artifacts: https://pubmed.ncbi.nlm.nih.gov/19874901/)

nicoa47, 2022-08-29:
moved to wiki instead --> close

closed 2022-08-29 by nicoa47