nicoa47/LDEEG #74: Pre-handler algorithms for specific datasets

caghangir-1, 2022-09-14:
def BCI_prehandler(file_path):
    
    raw = chetto_EEG.read_edf_file(file_path=file_path, preload=True) #example raw data
        
    # ====== Re-referencing ======
    raw_bip_ref = mne.set_bipolar_reference(raw, anode=['EMG_lower_left', 'EMG_lower_right', 'EOG_left', 'EOG_right'], 
                                            cathode=['EMG_ref', 'EMG_ref', 'EOG_right', 'EOG_left'], drop_refs=False)
    raw_bip_ref.drop_channels(ch_names=['EMG_lower_left', 'EMG_lower_right', 'EOG_left', 'EOG_right', 'EMG_ref'])
    # ====== Re-referencing ======
        
    #===== Pick info ======
    renamed_channels = {'EMG_lower_left-EMG_ref': 'EMG_lower_left', 'EMG_lower_right-EMG_ref': 'EMG_lower_right', 
                        'EOG_left-EOG_right': 'EOG_left', 'EOG_right-EOG_left': 'EOG_right'}
    channel_types = {'EMG_lower_left': 'emg', 'EMG_lower_right': 'emg', 'EOG_left': 'eog', 'EOG_right': 'eog'}
    
    try:
        raw_bip_ref.rename_channels(mapping=renamed_channels)
    except:
        raw_bip_ref.set_channel_types(mapping=channel_types)
    
    ch_names_final = raw_bip_ref.info['ch_names']
    ch_types_final = raw_bip_ref.get_channel_types()
    # #===== Pick info ======
    
    montage = mne.channels.make_standard_montage('standard_1005') #this is chosen
    raw_bip_ref.set_montage(montage) #final aim is to make it valid
        
    return raw_bip_ref

caghangir-1, 2022-09-14:
def BCI_prehandler_new(file_path):
    
    raw = chetto_EEG.read_brainvision_fil(file_path=file_path, preload=True) #example raw data
    
    # ======== Pre-handler ========
    raw.drop_channels(ch_names=['x7', 'x8'])
    channel_types = {'HEOG': 'eog', 'VEOG': 'eog', 'EMG1': 'emg', 'EMG2': 'emg', 'EMG3': 'emg', 'ECG': 'ecg'}
    raw.set_channel_types(mapping=channel_types)
    # ======== Pre-handler ========

    return raw

nicoa47, 2022-09-20:
moved [to wiki](https://github.com/nicoa47/LDEEG/wiki/Pre-handler-algorithms-for-specific-datasets).

closed 2022-09-20 by nicoa47